pschatzmann/logic-analyzer #1: ESP32 sample

hyoshiya2021, 2021-11-23:
According to your steps, I compailed and installed  the sample scetch for the esp32 board using Arduino IDE.
I can find new device on the pulseview's "Cennect to Device menu", and connecting sequence was successful.
Everything went well, I think.
When I clicked the run button on the pulseview screen, ESP32's monitor LED lights up few seconds and the sampling sequence go well.
But nothing(singnal monitor line) appears on the pulseview screen.
I monitored ESP32's Tx&Rx signals using another logic analyzer(using pulseview).
The Rx-signal(PC to ESP32 command reset,trigger etc.) wes succsessful.
The Tx-Signal(ESP32 to PC sampled start&data&stop) was succsessful.
What wrong?
I missed some settings?  Some steps being insufficient?
Please help me.

pschatzmann, 2021-11-25:
I have just committed a correction. 
Please note that pin 0 is GPIO19 in the example sketch.
Also note that a sampling rate >= 5MHz is also not giving any results, because this value is out of the limits of the ESP32...

hyoshiya2021, 2021-11-27:
Thanks for quick replay. I checked your correction and compiled&installed new one for esp32.
Pulseview's singnal monitoring function revived.
New trouble appeared on the pulseview's screen.
Signal monitor line changes regularly(on/off time 1:3 500kHz at 2Mhz sampling).*1
As before, external RX- Tx- signal monitor(using another logic analyzer) is succsessful(& signal structure is normal).
So I monitored Esp32's Tx-line using esp32's own GPIO19(seif-monitoring), in this case signal changes regularly as *1.
My conclusion is ... there is some trouble in the data processing before displaying sampling data on the pulseview's screen.
Rx- Tx- data transfer and data sampling are succsessful, but display data processing before displaying has some bags.
Best regards,

pschatzmann, 2021-11-27:
Do I understand you correctly that you see a signal even when nothing is connected ?

hyoshiya2021, 2021-11-27:
Thanks, I connected GPIO19 to TX- and GPIO22 to 3.3V Vcc.
External monitor TX data Is 10001000 00000000 00000000 00000000(01h 00h 00h 00h), correctry sampled I think.
But pulseview's display shows same data (on/off time 1:3 500kHz at 2Mhz sampling) on P0,P4 signal lines.
When I monitored Rx- TX-line at the same time, two lines display  the same(avobe) time transition.
My data checking sequense is blow:
 1) Waiting External Logic analyzer on Tx- trigger.(monitoring Rx- Tx lines at the same time)
 2) click ESP32's "run" button. ..external logic analyzer starts sampling(data sampled succsessfully)
 3) ESP32's self monitoring displays same on/off time transition.
I will check no input case as soon as possible, i can't remenber the display data on the no input case.
Maybe external logic analyzer's data is correct but ESP32 displays nothing(always 0-level).

pschatzmann, 2021-11-27:
The following simple tests would be interesting:
- Result when nothing is attached to pin19
- Result when pin19 is attached to ground
- Result if constant signal 3V is attached to pint19

You can also try to decrease the transmission speed by replacing SERIAL_SPEED  with 115200 in the sketch. (I never had any issues with 921600, but maybe this is too much for your environment)

hyoshiya2021, 2021-11-27:
According to SUMP protocol specification , my default SERIAL_SPEED is 115200.(115200 is max.??)
I wll check your advice now.　and will report.
thanks.

pschatzmann, 2021-11-27:
Just for your reference. I am usually using pwm to test because this is much easier:
Add the following to your sketch 
```
    ledcSetup(0, 5000, 8);
    ledcAttachPin(4, 0);
    ledcWrite(0, 128);
```
Connect pin4 to a input pin - I was using D21:

[image]

hyoshiya2021, 2021-11-27:
I checked your advice.
 1) pin19 signal level:
     open: no signal(low_level)
     ground: no signal(low_level)
     3.3v Vcc on/off : 1:3 patterns appears
2) your sample signal
     signal block pattern is same as you(above)
     signal in the block is on/off 1:3 pattern appers
conclusion:
 At any case of signal exist, display pattern is on/off 1:3 patten only.
 New evidence, changing the sample frequency :1:3 pattern is 1 sanple on 3samples off(checked 500khz,1MHz,2Mhz)
(1:3on/off pattern is not depend on sampling frequency)

pschatzmann, 2021-11-27:
Confirmed - I get the same.

pschatzmann, 2021-11-27:
I think I found the issue. Here is the result after the correction
[image]
:

hyoshiya2021, 2021-11-27:
Thanks for your quick response using valuable holiday.
I confirmed Pin19 signal-level check & your dummy signal check.
I think that issues are solved.
I'll check your corrected scetch operation in detail tomorrow holiday.
Pleae wait one another day for my checking in detail.
Additional questions:
 1) Esp32's display signal level are inverted compare to Saleae analyzer(or fx2lafw type analyzer)..<cheap chinease clone>why?
    (start is low_level, actually high_level is correct? I think)
 2) Esp32's(SUMP protocol) trigger kinds are only 3 compare to Saleae types(it has 6 kind of triggers).
     Can I set trigger kind to fall_down( rise-up) edge for esp32_analizer?
Best regards,

pschatzmann, 2021-11-27:
I did not put any effort into the trigger logic. As far as I remember just a parallel trigger is implemented where you can define for each line if it should be high or low. But honestly I have never tested this

hyoshiya2021, 2021-11-28:
I comfirmed that ESP32 logic analyzer works normally.  
Thanks for your serious and quick support.
The issues are solved.
I'll continue checking and confirming  your sketch for my skill up.
Best regards,

closed 2021-11-29 by pschatzmann