pschatzmann/logic-analyzer #2: Signals are transmitted "backwards" to PulseView

fedy2, 2022-01-24:
The signals values are transmitted "backwards" to PulseView.

PulseView expects the data to be in reverse order (OLS implementation of SUMP):
https://github.com/sigrokproject/libsigrok/blob/master/src/hardware/openbench-logic-sniffer/protocol.c#L499

Similar discussion in the [logic analyzer for Arduino](https://github.com/gillham/logic_analyzer): [issue #38](https://github.com/gillham/logic_analyzer/issues/38)

pschatzmann, 2026-09-12:
Fixed in 7aa0c6b. You were right — the sample bytes were being sent in network (big-endian) byte order via `htonl()`, but PulseView/libsigrok's OLS driver reconstructs each 4-byte sample as little-endian (`sample[0] | sample[1]<<8 | sample[2]<<16 | sample[3]<<24`). Since every board this library targets (ESP32, ESP8266, RP2040/RP2350, AVR) is little-endian, the fix is to just send the raw native byte order with no conversion.

While in there I also found that the single-sample `write()` used by continuous/streaming capture was calling `stream_ptr->write(htonl(bits))` — Arduino's `Print::write(unsigned long)` silently truncates to the low byte, so only 1 of the intended 4 bytes was actually being sent on that path. Fixed to write all 4 bytes explicitly.

Verified byte order and clean fixed-length captures against a Pico 2W with sigrok-cli. Please re-test and reopen if PulseView still shows reversed data.

pschatzmann, 2026-10-14:
Closing: the fix (native little-endian sample order) has been on main since 7aa0c6b. Please reopen if PulseView still shows reversed data.

closed 2026-10-14 by pschatzmann