pschatzmann/logic-analyzer #3: pico-pio problem

Eelkhoorn, 2022-02-24:
Compiled pico-pio and uploaded to raspberry-pico.
Pulseview seems to insert 3 extra (low) samples per sample.
Screenshot with GP6 to 3.3V:

[image]
.

pschatzmann, 2026-09-12:
Found the root cause: `PicoCapturePIO::arm()` configures the PIO state machine's input pins (`sm_config_set_in_pins`) but never actually routes the GPIOs to the PIO block or sets their direction — it was missing the `pio_gpio_init()` / `pio_sm_set_consecutive_pindirs()` calls that the [pico-examples logic_analyser.c](https://github.com/raspberrypi/pico-examples/blob/master/pio/logic_analyser/logic_analyser.c) this was based on relies on. Without that, the pins stay assigned to SIO and the PIO effectively samples nothing, which lines up with the spurious/low samples you were seeing.

Fixed in 7aa0c6b. Tested on a Pico 2W: flashed `logic-analyzer-pico-pio`, confirmed via `sigrok-cli` that the device is detected and produces clean, correctly-sized captures (previously the pins weren't wired into the PIO at all). Please retest with your setup and let us know if the extra samples are gone.

pschatzmann, 2026-10-14:
Closing: fixed in 7aa0c6b (missing `pio_gpio_init()` / `pio_sm_set_consecutive_pindirs()` plus sample byte order), verified on a Pico 2W. Please reopen if you still see extra samples.

closed 2026-10-14 by pschatzmann