pschatzmann/logic-analyzer #5: Any measurement pins grouped in the table

spairo1741, 2022-04-18:
- not every esp32 board has all the next pins. Is it realistic that active pins should be listed in the table?
- -------------------------------------
Pin 18 -> 1 => OK 
Pin 19 -> 10 => OK 
Pin 21 -> 1000 => OK 
Pin 22 -> 10000 => OK 
Pin 23 -> 100000 => OK 
Pin 25 -> 10000000 => OK 
Pin 26 -> 0 => ERROR
Pin 27 -> 0 => ERROR
------------------------------------
int arr[] = {18, 19, 21, 22, 23, 25, 26, 27};		// free pins for esp32 Wroom 32
...
for(int pin : arr) {
			//for (int pin = pinStart; pin < pinStart + numberOfPins; pin++) {
		pinMode(pin, OUTPUT);
    digitalWrite(pin, HIGH);
...
}

pschatzmann, 2022-04-20:
I think it is quite tricky to come up with an efficient generic solution that works on all platforms. 
I suggest that you just implement your own subclass of PinReader. There you could handle all the necessary bit shift operations...

pschatzmann, 2026-09-12:
This is a design constraint rather than a bug: `PinReader::readAll()` on every board reads a single contiguous block of GPIO register bits starting at `pinStart` (e.g. `REG_READ(GPIO_IN_REG) >> start_pin` on ESP32) with one shift — it can't gather an arbitrary/non-contiguous set of pins like `{18, 19, 21, 22, 23, 25, 26, 27}` into one bitmask that way, since some of those pin numbers aren't valid GPIOs at all on a Wroom-32 (24 and 28-31 don't exist; several others are strapping/flash pins), which is why you saw `ERROR` on non-contiguous entries.

Supporting arbitrary pin sets would need a real redesign of `PinReader` (a per-pin loop with a lookup table instead of a single shifted register read), which is a bigger change than a quick fix — happy to take a PR for it, but I don't want to make that change speculatively without a specific board/pin layout to validate against. In the meantime the practical guidance is: pick `START_PIN`/`PIN_COUNT` so the range is contiguous and entirely made of usable GPIOs for your specific board variant.

pschatzmann, 2026-10-14:
Closing as not planned: `PinReader` reads one contiguous block of GPIO register bits for speed, so non-contiguous pin sets would need a different (slower) design. Choose `START_PIN`/`PIN_COUNT` so the range covers usable consecutive GPIOs on your board. PRs that add an optional pin-list reader are welcome.

closed 2026-10-14 by pschatzmann