pschatzmann/logic-analyzer #6: it don't work at all.

shownb, 2022-05-05:
runs in a esp32 chip. all the pin connect nothing.
but show this. tried difference kinds of esp32.
[image]

peddamat, 2023-01-08:
Unfortunately I'm in the same boat.  Tried on an ESP-WROOM-32.

pschatzmann, 2026-09-12:
A few concrete bugs found and fixed in 7aa0c6b that would explain "nothing works at all" on ESP32:

1. **The `logic-analyzer-esp32` example didn't compile at all** on current esp32 Arduino core (3.x) — it includes `esp_int_wdt.h`, which no longer exists in current ESP-IDF and isn't even used anywhere in the sketch. Removed the dead include.
2. **Sample data was sent in the wrong byte order** — `htonl()` was applied to outgoing samples, but PulseView/libsigrok's OLS driver expects little-endian samples (see #2), so PulseView would receive garbage regardless of the actual signal.
3. **Off-by-one buffer overflow in the ring buffer**: the wraparound check used `>` instead of `>=`, so on some capture sizes it wrote/read one element past the end of the allocated array — this can corrupt memory and crash/hang the sketch unpredictably, which matches "shows nothing, all pins connect nothing."

I don't have ESP32 hardware here to verify directly (verified the equivalent fixes on a Pico 2W instead), but these are real, reproducible bugs independent of any specific board. Could you pull the latest `main`, reflash, and let us know if capturing now works?

pschatzmann, 2026-10-14:
Closing, since the bugs that would explain this (the uncompilable ESP32 example, wrong sample byte order, ring-buffer overflow) have been fixed on main since 7aa0c6b. Please reopen with details if it still doesn't work on the latest version.

closed 2026-10-14 by pschatzmann