pschatzmann/logic-analyzer #7: Arduino MEGA 2560 not working

joskfg, 2022-05-19:
I'm trying to get it working but PulseView is not detecting the device. I tried to change the start ping and count to 22 and 8 to use digital pins from 22 to 30 but it is still not detected.

I'm using the simple example:
`#include "Arduino.h"
#include "logic_analyzer.h"

using namespace logic_analyzer;  

int pinStart=START_PIN;
int numberOfPins=PIN_COUNT;
LogicAnalyzer logicAnalyzer;
Capture capture(MAX_FREQ, MAX_FREQ_THRESHOLD);

void setup() {
    Serial.begin(SERIAL_SPEED);  
    Serial.setTimeout(SERIAL_TIMEOUT);
    logicAnalyzer.begin(Serial, &capture, MAX_CAPTURE_SIZE, pinStart, numberOfPins);
}

void loop() {
    if (Serial) logicAnalyzer.processCommand();
}`

pschatzmann, 2022-05-20:
Difficult to say. Did you select the correct Serial Port in Pulseview with 9600 Baud ?

pschatzmann, 2026-09-12:
Found a real bug in the AVR pin reader (7aa0c6b): `PinReader::readAll()` was reading `PORTB`/`PORTD`, which are the *output latch* registers, not the actual pin state. For pins configured as `INPUT`, `PORTx` does not reflect the external voltage on the pin at all (it's only meaningful for `OUTPUT` pins, or to enable pull-ups on `INPUT`). It needs to read `PINB`/`PIND` instead — fixed.

Also added `avr` to `library.properties`'s `architectures` list — it was missing despite `config_avr.h` explicitly supporting AVR, which could cause tooling to flag the library as incompatible with the Mega.

I don't have a Mega here to verify directly, so I can't confirm this resolves the "PulseView doesn't detect the device" symptom specifically (that failure happens during the ID/handshake phase, before any pin is even read, so it may be a separate baud-rate/wiring issue — the default AVR config uses 9600 baud in `config_avr.h`, make sure PulseView/sigrok is configured to match). But the PORTx/PINx bug would definitely have produced wrong/no capture data even if detection succeeded. Could you retest with the latest `main` and let us know what baud rate you have configured in PulseView?

pschatzmann, 2026-10-14:
Follow-up in 74d444b: the AVR `PinReader` was hardcoded to the Uno/Nano register layout (PIND = pins 0-7, PINB = pins 8-13), which maps to completely different pins on the Mega 2560. There is now a Mega-specific reader: the default start pin is 22 (pins 22-29 → `PINA`), and A0-A7 (`PINF`) / A8-A15 (`PINK`) are supported too, with a `digitalRead()` fallback for any other start pin. Compiles for `arduino:avr:mega`.

For detection, remember the AVR default is 9600 baud (`SERIAL_SPEED`), so PulseView must use the same speed. Closing; please reopen if it still doesn't work with the latest main.

closed 2026-10-14 by pschatzmann