pschatzmann/logic-analyzer #8: maximum usefull sampling frequency for esp32 is 20Khz

RomanLut, 2022-06-05:
Maximum usefull sampling frequency for esp32 is 20kHz.
With frequencies above 20kHz, samples are captured with incorrect frequency, pulses became longer or shorter.
At 1Mhz, 500us pulse is captured as 1300us.

pschatzmann, 2026-09-12:
Some of the general fixes just pushed in 7aa0c6b (particularly the sample byte-order bug in #2, which caused corrupted data regardless of speed) may improve what you're seeing — worth retesting.

That said, for the generic (non-PIO) `Capture` class used on ESP32, the underlying sampling is a software bit-bang loop (`captureAllMaxSpeed`/`captureAll` calling `pin_reader_ptr->readAll()` per iteration, with `delayMicroseconds()` between samples below the max-speed threshold). At high rates (hundreds of kHz to MHz), loop overhead, `delayMicroseconds()` granularity/jitter, and interrupts (WiFi/BT stack, RTOS scheduler) on the ESP32 all eat directly into the sampling interval, which is exactly the kind of distortion you're describing (pulse widths off, e.g. 500us measured as 1300us). This isn't something that can be fully fixed in a software delay loop — it's inherent to bit-banged sampling on a general-purpose CPU with a multitasking RTOS underneath.

The Pico's `PicoCapturePIO` class avoids this entirely by using the RP2040/RP2350's PIO block + DMA for genuinely hardware-timed capture, which is why that path can reach much higher reliable rates. There currently isn't an ESP32 equivalent (it would need the ESP32's own hardware peripherals - e.g. I2S in parallel/camera mode - to do the same trick), so for now I'd treat something well under ~50-100kHz as the realistic reliable ceiling for the ESP32 software-loop capture path, not the theoretical `MAX_FREQ` value. If you want higher-rate ESP32 capture, that would need a hardware-timed capture backend as a new feature, which I'm not going to speculatively build without ESP32 hardware to validate it against — happy to guide/review a PR though.

pschatzmann, 2026-09-12:
Follow-up: pushed a first cut of a hardware-timed capture backend for the original ESP32 in 1272de9 — `CaptureESP32I2S` (new example: `logic-analyzer-esp32-i2s`), using the I2S0 peripheral's DMA-driven parallel "camera mode" instead of the bit-banged loop, similar in spirit to how `PicoCapturePIO` uses the RP2040's PIO+DMA.

Caveat: ESP32's I2S0 needs an *external* clock in this mode, so the sketch self-generates it via LEDC PWM on GPIO32 and expects a jumper wire back into GPIO34 (the clock input) — not a plug-and-play replacement for the software-loop path, but should get you real MHz-range sampling once wired.

Status: compiles, flashes, boots without crashing, and is correctly detected by PulseView/sigrok-cli over SUMP. I was not able to verify the actual capture data end-to-end (needed the loopback jumper, wasn't available this session), so treat this as experimental until someone can confirm captured waveforms look correct with a real signal + the jumper in place. No triggering support yet either.

pschatzmann, 2026-10-14:
Closing: the bit-banged `Capture` loop is inherently limited on ESP32, so there are now hardware-timed DMA backends for high sample rates: `CaptureESP32I2S` for the original ESP32 (example `logic-analyzer-esp32-i2s`, 1272de9) and an LCD_CAM/DVP backend for ESP32-S3/P4 (example `logic-analyzer-esp32s3-lcdcam`, 7b16ea9). See docs/esp32.md and docs/esp32s3.md. Please reopen if you hit timing problems with those.

closed 2026-10-14 by pschatzmann