pschatzmann/logic-analyzer #12: Pi Pico PIO Example does nothing

zoggins, 2023-10-31:
I have done everything that I think is correct, but there is just no data coming out of the serial port.  Any help would be appreciated.  Thanks.

pschatzmann, 2023-10-31:
I haven't used this for a long time. 
Did you follow [these steps](https://github.com/pschatzmann/logic-analyzer#connecting-to-pulseview) ?

zoggins, 2023-10-31:
Yes.  I have done those instructions on multiple Picos and multiple computers and the device never shows up when i hit scan.  The Pico seems to be a some stuck state because the Arduino UI can't even force it to programming mode, I have to manually reboot the board to program it again.  I have also tried with the  testData function and real signals, all the same result.

zoggins, 2023-10-31:
Its no big deal I will either debug it myself or try something else.

zoggins, 2023-10-31:
It appears its something with the serial output.  I can hit it directly with a terminal for a little bit and then the serial communication will lock up.

pschatzmann, 2023-10-31:
Did you try to reduce the Serial speed ?

pschatzmann, 2026-09-12:
This should now be fixed by 7aa0c6b — same root cause as #3: `PicoCapturePIO::arm()` never called `pio_gpio_init()` / `pio_sm_set_consecutive_pindirs()`, so the PIO block wasn't actually connected to the GPIO pins, meaning the capture always read nothing regardless of the signal on the pins.

I also fixed:
- the SUMP sample byte order (was being sent big-endian instead of the little-endian PulseView/libsigrok expects, see #2)
- the `logic-analyzer-pico` example's `#ifndef PICO` guard, which isn't defined on Pico 2/2W boards (use `ARDUINO_ARCH_RP2040`)

Tested end-to-end on a Raspberry Pi Pico 2 W: compiled and flashed `logic-analyzer-pico-pio`, and `sigrok-cli --driver ols:conn=/dev/ttyACM0 --scan` now detects the device and captures the requested number of samples cleanly. Please pull the latest main and retest on your Pico — let me know if it's still not producing data.

pschatzmann, 2026-10-14:
Closing: same root cause as #3, fixed in 7aa0c6b and verified end-to-end with sigrok-cli on a Pico 2W. Please reopen if the PIO example still produces no data.

closed 2026-10-14 by pschatzmann