pschatzmann/logic-analyzer #13: Arduino UNO not working

vks007, 2025-01-05:
I have set up everything as in the intructions but on clicking the "scan" on pulview nothing comes up. I am using 9600 baud rate as in the example sketch. 
How can I verify that the ardunio is working fine , is it only via Pulseview or should I see something on the serial port?
[image]

This is the sketch I have uploaded:
```/**
 * @file logic_analyzer.ino
 * @author Phil Schatzmann
 * @copyright GPLv3
 * @brief Arduino Sketch for the sigrok LogicAnalyzer for the ESP32 and AVR processors using the SUMP protocol
 * See https://sigrok.org/wiki/Openbench_Logic_Sniffer#Short_Commands * 
 */

#include "Arduino.h"
#include "logic_analyzer.h"

using namespace logic_analyzer;  

int pinStart=START_PIN;
int numberOfPins=PIN_COUNT;
LogicAnalyzer logicAnalyzer;
Capture capture(MAX_FREQ, MAX_FREQ_THRESHOLD);

// Use Event handler to control the LED
void onEvent(Event event) {
    if (event == logic_analyzer::STATUS) {
        switch (logicAnalyzer.status()) {
            case ARMED:
                digitalWrite(LED_BUILTIN, LOW);
                break;
            case STOPPED:
                digitalWrite(LED_BUILTIN, LOW);
                break;
        }
    }
}

void setup() {
    Serial.begin(SERIAL_SPEED);  
    Serial.setTimeout(SERIAL_TIMEOUT);

    logicAnalyzer.setEventHandler(&onEvent);
    logicAnalyzer.setDescription(DESCRIPTION);
    logicAnalyzer.begin(Serial, &capture, MAX_CAPTURE_SIZE, pinStart, numberOfPins);
}

void loop() {
    if (Serial) logicAnalyzer.processCommand();
}```

pschatzmann, 2025-01-06:
If the serial port is not showing up, maybe it is still in use by another program (e.g. Arduino)

vks007, 2025-01-06:
The serial port is showing up but the device doesn't show up. I tried uploading the program on an ESP32 too and the same result. 
How should I debug further?

vks007, 2025-01-06:
I tried Arduino again and I was able to see the device. But my joy was short lived. It didn't show any data on running the session but threw back a "generic error" , I restarted Pulseview and now again it is not able to find the UNO device.

vks007, 2025-01-06:
I am closing the issue as it seems to be working now. Its very erratic though , anyway might have to play around more. At times when it works I have to unplug the Ardunio / ESP32 , restart Pulseview a couple of times and it suddenly starts working.

closed 2025-01-06 by vks007